BerkeleyLab/Marble-Mini #2: White Rabbit oscillators

gkasprow, 2018-12-04:
@michallgaska please follow recent Metlino/Sayma/RFSOC discusions about WR implementation.
We want to use same approach in all our designs. It was  demonstrated that with 2 I2C controlled XO and one clock buffer we cn achieve very good phase noise.
[Here ](https://github.com/sinara-hw/sinara/issues/515#issuecomment-416553527)is the discussion and results

ldoolitt, 2018-12-04:
On Tue, Dec 04, 2018 at 10:43:15AM -0800, Greg Kasprowicz wrote:
> We want to use same approach in all our designs. It was  demonstrated that with 2 I2C controlled XO and one clock buffer we cn achieve very good phase noise.
> [Here ](https://github.com/sinara-hw/sinara/issues/515#issuecomment-416553527)is the discussion and results

Has anyone checked the Si549 for Helium sensitvity?
https://hackaday.com/2018/10/31/helium-can-stop-your-iphone-maybe-other-mems-too/
https://www.youtube.com/watch?v=vvzWaVvB908
The problematic SiTime devices are low-frequency low-power devices, which sound more susceptible.
Still, maybe worth checking, especially since helium is not exactly rare in the target environments.

gkasprow, 2018-12-04:
I've read about helium issues with Iphones.
But this affects only MEMS devices.

ldoolitt, 2018-12-04:
On Tue, Dec 04, 2018 at 12:03:12PM -0800, Greg Kasprowicz wrote:
> I've read about helium issues with Iphones.
> But this affects only MEMS devices.

Yes, it's peculiar to MEMS.  Si549 is crystal-based.  Never mind.

ldoolitt, 2019-04-04:
Thanks @gkasprow for bringing this up, and providing the link to WR oscillator work!

I'm going to close this issue, for the following reasons:
1. We have a (presumably) working design based on the previous generation of WR: 25 MHz and 20 MHz VCXOs driven by DACs, and a CDCM61004 multiplying 25 MHz up to 125 MHz MGTREFCLK.
2. The new approaches are still in evaluation, and might end up dropping one device in favor of using internal Xilinx PLLs.
2. It's not clear to me that the new approach is consistent with the trust model assumed on this board.

This topic can be revisited after our first hardware iteration.

closed 2019-04-04 by ldoolitt