BerkeleyLab/Marble-Mini #3: Consider table-based entry of pin mapping

ldoolitt, 2019-01-10:


jeinstei, 2019-01-10:
migen/litex style?

ldoolitt, 2019-01-10:
See attached table.tar.gz.

Using a 60-line python program and some _extremely_ simple lists
of application signals, a netlist file (429 entries) is generated.
With some extra work, I assert this netlist can be injected into KiCad.
I'd need to see the KiCad files as you have them now, before I can work
on that step.

The result would IMHO be far more legible and useful going forward than
having page after page of dense signal lists entered into a GUI.
E.g., pins_for_review.png is a screenshot of "paste b*.txt | less -x15" 
that captures all of the FPGA user I/O mapping for this board.
Note that the key data file xc7a100tfgg484pkg.txt is directly downloaded
from Xilinx.

[table.tar.gz](https://github.com/gkasprow/Simple_AFC/files/2747238/table.tar.gz)

[image]

ldoolitt, 2019-01-10:
@jeinstei: I'd take anything based on flat files.  The example I just posted is the simplest I can imagine, but if migen/litex has something similar --- and comprehensible by mortals!  --- I'd take that too.

ldoolitt, 2019-01-10:
@jeinstei: I've messed around with Litex some.  I respect bunnie, and the blog posting that you reference was part of the kick I needed to get started.  My first-hand experience with Litex is what made me add the caveat "comprehensible by mortals".

ldoolitt, 2020-03-05:
I can't see anyone putting in effort to convert this working design to a different design paradigm, so I'll close this issue.
I still think that designs of this scale (and larger) would greatly benefit from using table-based entry of pin mapping. When scripts have access to this information, it's easy and natural to generate cross-reference tables and other derived files like FPGA constraints.

closed 2020-03-05 by ldoolitt