BerkeleyLab/Marble-Mini #5: Add WR-CLK compatibility

kaolpr, 2019-03-20:
At Creotech we're developing a WhiteRabbit CLK module. It is a MCH tongue that allows distribution of WR clock, PPS and timestamp to all AMC modules in a crate without any external cabling (besides fiber to the WR-CLK).

I see that it is possible to make Marble compatible with WR-CLK with relatively low effort:

* TCLKA and TCLKB should be connected to FPGA (SRCC pins),
* at least two MLVDS RX pairs should be connected (ports 17, 18).

Required pins are available in banks 15 and 16.

ldoolitt, 2019-03-21:
> At Creotech we're developing a WhiteRabbit CLK module.
> [It] allows distribution of WR clock [and data].

Sounds great!

> I see that it is possible to make Marble compatible with WR-CLK with relatively low effort:
> Required pins are available in banks 15 and 16.

Maybe.

> * TCLKA and TCLKB should be connected to FPGA (SRCC pins),

TCLKA and TCLKB are diff pairs on AMC backplane.
CC pins in general are a precious commodity in this design.  Let me summarize their allocation per-bank (commit 9dde4030 or c8fd3271):
  b13.  FMC2 LA17 LA18, 2 non-clock; 35-pin bank, half of FMC2, no spare pins
  b14.  FMC1 LA0 LA1 CLK0 CLK1; full
  b15.  MII_RX_CLK, FMC1 LA17 LA18, 1 non-clock; LA33 could be moved
  b16.  VCXO, TMDS, 2 non-clock; 3.3V bank, not valid for LVDS output
  b34.  FMC2 LA0 LA1 CLK0 CLK1; full
  b35.  DDR3 only; full

I can see how to add one LVDS clock to bank 15; there are (more than) two empty pins, and LA33 can be moved.  Will require a remap of at least one MII pin, e.g., MII_RX_DV to H15.

Adding the second LVDS clock is harder; all I can think to do is displace MII_RX_CLK to b16.  That sounds like it will mess up both its timing and voltage threshold compared with all the other MII pins.

When counting b15 pins in the current design, you can consider PHY_RSTn = M17 available after a remap; it's not timing critical. 

The other approach is to use pairs on b16, with its 3.3V bank supply.  That prevents use of internal termination, so a 100 Ohm external resistor would be needed per input pair.

> * at least two MLVDS RX pairs should be connected (ports 17, 18).

It sounds like you're asking for two more LVDS input pairs to come from the AMC backplane.  Do these inputs need to be on the same bank as TCLKA and TCLKB?  Are there special considerations for MLVDS on Xilinx 7-series HR pins, distinct from LVDS?  I'm clearly not an xTCA expert; I don't see identifiable "ports 17, 18" on the AMC fingers in our schematic.

Eight total pins are not available on b15 without totally rethinking the MII interface.  If, instead, we consider using b16 and external termination resistors, six pins are immediately available, including two clock pairs.  Freeing up two more pins there will take a little effort, but a couple low-speed signals could be moved to a 2.5V bank.  Worst case, that means adding new level-shift chips.

If the LVDS inputs and the clocks can be on different banks, things might get a bit more flexible.  I have not looked at the layout to see if that adds extra constraints; I know b14 is pretty much surrounded.

ldoolitt, 2019-03-21:
Mikołaj -

Thanks for looking at our board design!

On Wed, Mar 20, 2019 at 04:24:36AM -0700, Mikołaj Sowiński wrote:
> At Creotech we're developing a WhiteRabbit CLK module.
> [It] allows distribution of WR clock [and data].

Sounds great!

> I see that it is possible to make Marble compatible with WR-CLK with relatively low effort:

Maybe.

> * TCLKA and TCLKB should be connected to FPGA (SRCC pins),

TCLKA and TCLKB are diff pairs on AMC backplane.
CC pins in general are a precious commodity in this design.  Let me 
summarize their allocation per-bank (commit 9dde4030 or c8fd3271):
  b13.  FMC2 LA17 LA18, 2 non-clock; 35-pin bank, half of FMC2, no spare pins
  b14.  FMC1 LA0 LA1 CLK0 CLK1; full
  b15.  MII_RX_CLK, FMC1 LA17 LA18, 1 non-clock; LA33 could be moved
  b16.  VCXO, TMDS, 2 non-clock; 3.3V bank, not valid for LVDS output
  b34.  FMC2 LA0 LA1 CLK0 CLK1; full
  b35.  DDR3 only; full

I can see how to add one LVDS clock to bank 15; there are (more than)
two empty pins, and LA33 can be moved.  Will require a remap of at least
one MII pin, e.g., MII_RX_DV to H15.

Adding the second LVDS clock is harder; all I can think to do is displace
MII_RX_CLK to b16.  That sounds like it will mess up both its timing and 
voltage threshold compared with all the other MII pins.

When counting b15 pins in the current design, you can consider
PHY_RSTn = M17 available after a remap; it's not timing critical.

The other approach is to use pairs on b16, with its 3.3V bank supply.
That prevents use of internal termination, so a 100 Ohm external
resistor would be needed per input pair.

> * at least two MLVDS RX pairs should be connected (ports 17, 18).

It sounds like you're asking for two more LVDS input pairs to come
from the AMC backplane.  Do these inputs need to be on the same bank 
as TCLKA and TCLKB?  Are there special considerations for MLVDS on
Xilinx 7-series HR pins, distinct from LVDS?  I'm clearly not an xTCA 
expert; I don't see identifiable "ports 17, 18" on the AMC fingers in
our schematic.  

Eight total pins are not available on b15 without totally rethinking
the MII interface.  If, instead, we consider using b16 and external
termination resistors, six pins are immediately available, including two 
clock pairs.  Freeing up two more pins there will take a little effort,
but a couple low-speed signals could be moved to a 2.5V bank.  Worst 
case, that means adding new level-shift chips.

If the LVDS inputs and the clocks can be on different banks, things
might get a bit more flexible.  I have not looked at the layout to see 
if that adds extra constraints; I know b14 is pretty much surrounded.

  - Larry

kaolpr, 2019-03-25:
Sorry far such a late answer.

For minimal compatibility with WR-CLK the following is needed:

- TCLKA -> CC diff pair (this line delivers WR clock)
- 2 MLVDS RX diff pairs (one pair for PPS, other for timestamp)

No restrictions on bank placement.

gkasprow, 2019-03-25:
for MLVDS please use dedicated MLVDS transceivers. Standard LVDS would form long stubs. MLVDS transceivers will also enable using 3V3 LVCMOS lines.

ldoolitt, 2019-03-25:
OK, this is starting to sound plausible.  One possible approach:

Move FMC1_LA33 away from a IO_11(SRCC) pair, still within b15.  This will require a cascade of displacements, depends some on routing; one scenario (ignoring routing):
- MII_RX_DV to IO_L14N (pair with MII_RX_CLK)
- FMC1_LA33 to IO_L20,
- Attach TCLKA to IO_11(SRCC).

Add two channels of MLVDS receiver, output to any two convenient FPGA user I/O pins.
- 1 pin free on 2.5V b15, doesn't sound like a good choice
- 6 pins free on 3.3V b16

I need to try a test synthesis to confirm that it's OK to use IO_L14N as data at the same time that IO_L14P is used as a (single-ended) clock.

I'll defer to Michal whether he thinks this can/should be put in for the first article, or if it would add too much time/effort to the schedule.

ldoolitt, 2019-03-25:
@kaolpr:
>  2 MLVDS RX diff pairs (one pair for PPS, other for timestamp)

Couldn't a single lane cover this?  GPS receivers with RS-232 NMEA outputs do this all the time. The start of the timestamp message _is_ the PPS marker.

kaolpr, 2019-03-28:
@ldoolitt Well, it can be done. Let's reduce compatibility requirement to:

- single diff pair for TCLKA,
- single RX MLVDS pair.

gkasprow, 2019-05-02:
@michallgaska I see that you did not add support for WR. This is simple - add MLVDS transceiver and TCLK.

michallgaska, 2019-05-27:
Done.
Can be closed.

ldoolitt, 2019-05-27:
There is no value to controlling U42-2 (#RE), just tie it low.
That way we can all see that FPGA pin C17 (IO_L12N_T1_MRCC_16) is still unused.

Please name the R net (U42-3) to something short and descriptive.
This will turn into the FPGA pin name in the HDL design.

ldoolitt, 2019-05-31:
Confirmed fixed as of commit 0fbafa19

closed 2019-05-31 by ldoolitt