BerkeleyLab/Marble-Mini #8: schematics cosmetics

gkasprow, 2019-03-22:
cosmetic issues
- [ ]  global FMC labels should be bi-dir not only inputs
- [ ] in all OHWR designs we use certain power port nomenclature like P3V3, N5V0 etc. Is there some reason this was changed? @ldoolitt any preference here
- [ ] too many 2k2 I2C pullups. Make sure that when certain port is enabled, the pullup resistance is not lower than 1k. THe switch adds some R to the path and may cause issues. I recommend replacing all pullup R after the switch with 22k
- [ ] the Q3 seems to be installed in wrong direction. The inrush current limiter will be bypassed by internal diode.
- [ ] please add I2C devices map t othe I2C switch sheet to make sure we don't have address overlaps
- [ ] for SFP cages please set right Rd/Td line directions. Now are bidir. RD is output, TD is input.
- [ ] add brief name/description of the schematic in the largest frame field.
- [ ] for some unknown reason the circuit protecting against glitches on MMC ISP was removed. I suggest adding at least the switch that disables that function when not needed. Otherwize on some PCs the MMC will enter boot mode when USB is plgged and  DTR set to low.

ldoolitt, 2019-03-22:
> in all OHWR designs we use certain power port nomenclature like P3V3, N5V0 etc. Is there some reason this was changed? @ldoolitt any preference here

No preference beyond self-consistency in this design.

> too many 2k2 I2C pullups. Make sure that when certain port is enabled, the pullup resistance is not lower than 1k. The switch adds some R to the path and may cause issues. I recommend replacing all pullup R after the switch with 22k

By "switch" do you mean U5 (TC9548)?  That's a good point; I always looked at that as an active component, but it's actually passive, in a way.  I agree with your 22k solution.  Or maybe 3.3k upstream, 10k downstream?

> please add I2C devices map to the I2C switch sheet to make sure we don't have address overlaps

The I2C subsystem is indeed complicated, and needs better documentation. I was working on that just now.  Address overlaps are not actually a problem, at least after Issue #6 item 1 is fixed.

ldoolitt, 2019-03-23:
> The I2C subsystem is indeed complicated, and needs better documentation.

Here's my first try.  The source eps (xcircuit) file can be supplied on request.
Applies to commit de3cfc8 assuming Issue #6 item 1 is fixed.
Note bus address conflict for U34 and U39, that also needs fixing.
I2C bus signals are named I2C_${BUS}_{SCL,SDA} on the schematic.

[image]

ldoolitt, 2019-04-04:
@gkasprow please split this list into "bug" (must be fixed before first fab) and "enhancement" (OK to delay work until future revision).  If everything is in the "bug" category, just mark this issue as such.

gkasprow, 2019-04-11:
we can close it now, remaining issues are not relevant anymore.

ldoolitt, 2019-04-15:
Checking commit 38e794c1.  Remaining items in order of importance:

> for SFP cages please set right Rd/Td line directions. Now are bidir. RD is output, TD is input.

SFP1 and SFP2 are fixed, but SFP3 and SFP4 are still wrong.  Please fix.

> add brief name/description of the schematic in the largest frame field.

Not added yet.  Definitely in the "nice to have" category.

> power port nomenclature

Can be left as is for prototyping; we'll see what a larger audience says later.

ldoolitt, 2019-05-02:
Closing this issue.  Thanks @michalgaska for following up on seven of the sub-issues!
If we get complaints about the power supply names, we can update after the first prototypes are made and tested.

closed 2019-05-02 by ldoolitt