BerkeleyLab/Marble-Mini #9: Power objects +3V3 and +3V3MP

michallgaska, 2019-03-26:
For some reasons, KiCad can't separate +3V3 and +3V3MP on PCB. It is seen as the same object on PCB. I thought that it was my mistake when I created the power object in the library. But I have tried a few times and I still have the same issue.  
Due to this issue, I can't finish two polygons.
I asked Tom Włostowski for a piece of advice in email but I still have been waiting for his answer.

gkasprow, 2019-03-26:
Just change the name of one of them to i.e. P3V3MP Aren't they shorted somewhere on the schematics? Can you look to the netlist if they are not connected somewhere?

gkasprow, 2019-03-26:
@michallgaska U10 and U7 have hidden net which is connected to +3V3MP. It may have already assigned net name to supply pin and this can cause the issue

michallgaska, 2019-03-27:
I found it.
Some devices have a supply pin named VCC and for an unknown reason, KiCad connects these pins to +3V3.
I changed pins name and it solved the issue.

closed 2019-03-27 by michallgaska