BerkeleyLab/Marble-Mini #13: DC drop simulations

michallgaska, 2019-04-08:
I made a preliminary simulation of voltage drop on power lines.
There are no huge voltage drops so it looks very promising.
Polygons have slightly too big clearance to bias in comparison to the original project.
Results would be a little better if I had Hyperlynx files with proper clearance. 

**1V @ 6A:**
[image]

**1V MGTAVCC @ 0.2A**
[image]

**1.2V MGTAVTT @ 0.1A**
[image]

**1.35V @ 0.55A**
[image]

**1.8V VCCAUX @ 0.4A**
[image]

2.5V @ 1.1A
[image]

3.3V and 3.3MP 
TODO

gkasprow, 2019-04-08:
@michalgaska did you update the stackup in Hyperlynx? The tool places default stackup since the Kicad does not have the stackup editor.

michalgaska, 2020-09-12:
Simulations were made with proper stackup. 
The issue can be closed.

closed 2020-09-12 by michalgaska