BerkeleyLab/Marble-Mini #15: ERC errors

ldoolitt, 2019-04-15:
As of commit 38e794c1, schematic ERC (Electrical Rules Check?) reports 10 Errors and 423 Warnings.  I'm not qualified to comment on whether any of these represent actual netlist errors that need to be fixed before prototyping.  I do think the design should pass ERC before it can be considered done and releasable.

ldoolitt, 2019-04-26:
As of commit fdb63ea5, schematic ERC reports 11 Errors and 53 Warnings.  Sounds like an improvement to me, thanks @michalgaska!  It would be nice to get the Error count down to zero, if that's possible.

yetifrisstlama, 2019-04-27:
some thoughts on the schematic ERC errors I was seeing:

  * `Label not connected anywhere else in the schematic` I've seen this on many busses. Looks like a Kicad bug / weirdness to me, as the highlight net tool shows the connection is there
  * `Pin connected to other pins, but not driven by any pin`  ... needs a PWR_FLAG symbol placed on the net to tell Kicad it is driven by a power source
  * `Pin not connected (use a "no connection" flag to suppress this error)` ... self explaining
  * `More than one name given to this bus or net` ... i suspect this is due to a hidden pin of the same name. Not sure how to solve this

ldoolitt, 2020-05-05:
Fixed (to the maximum extent possible) in commit 6cafe828 (branch fab2).
Still one unavoidable ERC error due to the ERC process not understanding DNF attribute of Y1.

closed 2020-05-05 by ldoolitt