BerkeleyLab/Marble-Mini #18: Integrate KiCad workflow better with git

ldoolitt, 2019-04-15:
At the minimum, I think the following files should be removed and listed in .gitignore:
* AMC_FMC_Carrier-PcbDoc.kicad_pcb-bak
* fp-info-cache
* AMC_FMC_Carrier-PcbDoc-rescue.lib

but please check this with git and/or KiCad experts, and test that the resulting repo is usable.

There's a longer set of instructions for using git and KiCad together at [plotkicadsch](https://jnavila.github.io/plotkicadsch/).  There are clearly some good lessons in there, but in total it's kind of intricate. We don't have to do everything all at once.

ldoolitt, 2019-05-16:
@michalgaska and I have both made commits addressing this issue.

Other people should check that the result is usable, and maybe leave a comment here.

yetifrisstlama, 2019-06-03:
Looks good to me, can be closed.

ldoolitt, 2019-06-04:
Closing now.  Please re-open if you (anyone) find specific problems.

closed 2019-06-04 by ldoolitt