BerkeleyLab/Marble-Mini #19: Logos

ldoolitt, 2019-04-16:
Add as many as you can think of: Creotech, Politechnika Warszawska, LBNL, KiCad, OHWR.
Also designers' names: Michal, Larry, and as many others as want to be associated with the result.
If you can find room, getting something in copper (not just silkscreen) is a "nice to have".
Directly before release, add a revsion date and git commit id.

ldoolitt, 2019-04-29:
The logos are on and look great!  Thanks @michalgaska!
Leaving this issue open as a reminder to
> Directly before release, add a revision date and git commit id.

csp, 2019-04-29:
I see that the front has two areas with logos, one of which will be covered when a mezzanine is connected. For the others, can we replace those with at least LBNL and Creotech logos?

gkasprow, 2019-05-02:
I'm not sure if we can place CE logo here. We did not certify this board.
The board is just a component so we would rather certify the whole box with this board and place the CE logo there.

gkasprow, 2019-05-02:
We usually do not place individuals names on the boards. It's a collective effort so just institution name is sufficient. Names are placed on schematics and PCB drawing tables where indicate who designed and who verified each part.

ldoolitt, 2019-05-02:
I agree a board like this is a big team effort - and putting more names on is fine.
But if you're uncomfortable with people's names on the board, take them off.
We at LBNL typically do put the designers' names on.  I think of it like an artist signing a painting, giving credit and recognition to someone who put in the effort.

gkasprow, 2019-05-02:
well, it's not really a piece of art :) It's good engineering. I had a colleague who treated PCBs like a piece of art but they usually did not work well because for him good look was more important than Signal Integrity :)
On commercially available electronics you won't find authors names. 
A long time ago we had such discussion at OHWR and the conclusion was that we do not place a list of all contributors, but a link to the repo where this information is provided.

ldoolitt, 2019-05-02:
> On commercially available electronics you won't find authors names.

One exception is on chip masks, where you find all kinds of crazy stuff.  :-)
I can accept the conclusion, no names on this board.

ldoolitt, 2019-05-10:
Closing because it's mostly done, and the un-done stuff is not related to the issue title.
Will open a fresh issue for revision date and git id.

csp, 2019-05-30:
Re-opening this issue. This is my original request regarding logos: I see that the front has two areas with logos, one of which will be covered when a mezzanine is connected. For the others, can we replace those with at least LBNL and Creotech logos?

ldoolitt, 2019-06-03:
Confirmed good in commit 0293096. Thanks @michalgaska!

closed 2019-06-03 by ldoolitt