BerkeleyLab/Marble-Mini #20: Please provide FPGA pin mapping file

ldoolitt, 2019-04-22:
Either simple flat-text, or .xdc for Vivado.
Ideally this will machine-generated from the KiCad schematic, with instructions for doing so in the README (hope for "make marble.xdc").
If some post-processing is needed to get this into a useful form, we at LBNL can help.
It's not definitely not acceptable to key it all in by hand, A really ugly but possible short-term workaround is to cut-and-paste things bank-by-bank from the PDF.

gkasprow, 2019-04-23:
I used to extract such info from raw netlist. It's just a metter of copying
piece of netlist that contains nets around FPGA designator. One remark -
make sure all FPGA nets are named. Sheet entries do not alway name nets so
I always add net names close to the FPGA pins

ldoolitt, 2019-04-23:
@gkasprow wrote: I used to extract such info from raw netlist. 

Sure, there's always a way.  The more automated and guaranteed-correct the better.
The path would be more clear if anyone had followed through on #3.

gkasprow, 2019-05-02:
Orcad PCB netlist output seems to be human-readable form. It's the only format among supported by Kicad where nets are grouped by symbols.
The netlist looks almost ready to use :
```

(  AA3 FMC2_LA_9_N )
  (  AA4 FMC2_LA_1_N )
  (  AA5 FMC2_LA_15_P )
  (  AA6 FMC2_LA_11_N )
  (  AA7 +2V5 )
  (  AA8 ? )
  (  AA9 FMC2_LA_27_P )
  ( AA10 FMC2_LA_20_P )
  ( AA11 FMC2_LA_20_N )
  ( AA12 GND )
  ( AA13 FMC2_LA_31_P )
  ( AA14 FMC2_LA_21_N )
  ( AA15 FMC2_LA_32_P )
  ( AA16 FMC2_LA_26_N )
  ( AA17 +2V5 )
  ( AA18 FMC1_LA_6_P )
  ( AA19 FMC1_LA_10_P )
  ( AA20 FMC1_LA_7_P )
  ( AA21 FMC1_LA_7_N )
```

after one minute of play with text file I got ready to use pin mapping
[FPGA_pin_mapping.txt](https://github.com/gkasprow/Simple_AFC/files/3137409/FPGA_pin_mapping.txt)

ldoolitt, 2019-05-10:
This is helpful, and I'll add a separate issue for myself to synthesize a test bitfile.
I will leave this issue open until there's a fully automatic flow from KiCad to Xilinx .xcd file.

ldoolitt, 2020-03-05:
This work was completed August 2019 with commit a1ed1ff.
Instructions for how to generate Marble.xdc are in the README.
Tested on hardware, works.
Thanks, @michalgaska!

closed 2020-03-05 by ldoolitt