BerkeleyLab/Marble-Mini #21: Improve support for FMC JTAG

ldoolitt, 2019-04-25:
If I read the schematics correctly, the only access provided to the FMC JTAG pins is via J14 Conn_ARM_JTAG_SWD_10.  The JTAG chain (slaves) accessible from that connector starts with U2 LPC1776, and will have FMC1 and FMC2 switched in when the corresponding PRSNT_M2C signals are asserted.

My understanding is that a few FMC mezzanine cards (including CUTE-WR) use their JTAG lines for maintenance. While these boards could be tested with the current design of Marble, by attaching a JTAG pod to J14, there's no sane way to support deployment of such cards.

Please add the capability to drive these FMC JTAG pins by GPIO pins of U2 LP1776 (MMC), which can bit-bang the JTAG protocol.  The MMC is presumed to be attached (or attachable, however indirectly) to some source of programming instructions for the FMC mezzanines. You can keep J14's capability to master FMC JTAG as a resistor-strappable option.

This concern was prompted by studying ERC output, in combination with emails from last July.
The current design leaves FMC TRST pins floating; shouldn't they be tied down somehow?

michallgaska, 2019-04-26:
I connected both FMC JTAG to MMC.
Driving FMC JTAG by MMC should be a default mounting option?

gkasprow, 2019-04-26:
I'm not sure if I follow you. The idea behind FMC JTAG was identical as we do in Sayma AMC. Some FMC modules, like CUTE-WR nodes, have FPGA on FMC. The FPGA JTAG is accessible via FMC JTAG.
So, once the FMC is inserted, the FMC is included in the carrier JTAG chain.
There is a risk that FMC is not designed according to specification and the chain is not closed by default (TDO not shorted with TDI). In such a case the MMC should be able to overwrite the PRSNT pin and close the FMC JTAG chain by the analog switch.
Xilinx devkits have JTAG implemented in the same manner.
MMC can use bit-bang JTAG, there is even Xilinx virtual cable protocol, but it makes sense when MMC has Ethernet access. I don't see many use cases for MMC- FMC JTAG connectivity.

ldoolitt, 2019-04-26:
> I'm not sure if I follow you.

I agree this is confusing, and that's bad.  For a start, I request more notes on the schematic about which zero-ohm resistors are placed to enable which features.

> The idea behind FMC JTAG was identical as we do in Sayma AMC.

OK.  I haven't looked at Sayma design in months, I guess i should refresh myself.
Can a remote in-the-field Sayma carrier, not attached to a bench-top JTAG pod, act as a JTAG bus master for the FMC JTAG chain?

> Some FMC modules, like CUTE-WR nodes, have FPGA on FMC.

Yes.  Clarifying, some FMC mezzanine cards with an on-board FPGA attach that chip's dedicated JTAG slave port to the FMC connector's dedicated JTAG pins.

> There is a risk that FMC is not designed according to specification and the chain is not closed by default (TDO not shorted with TDI). In such a case the MMC should be able to overwrite the PRSNT pin and close the FMC JTAG chain by the analog switch.

Right, that's a useful feature.  I guess it is part of the Sayma design.  Is it part of our Marble design now?

> Xilinx devkits have JTAG implemented in the same manner.

Can you be a bit more specific?  Which devkit? Are you talking about the analog switch with an override feature? And is this the JTAG chain that is only accessible to a bench-top JTAG pod?

> MMC can use bit-bang JTAG, there is even Xilinx virtual cable protocol, but it makes sense when MMC has Ethernet access.

The MMC is always assumed to have a data link to the outside world.  In xTCA that's over IPMB.  In the NAD world I will set up a bridge from Ethernet through the FPGA via SPI.

> I don't see many use cases for MMC- FMC JTAG connectivity.

OK.  But I don't want to force people to add extra hardware in order to access the FMC JTAG chain.  That hardware may be easy and convenient in a lab setting, and that use case should be supported, but Marble is intended to support deployment of tens or hundreds of units out in the field where physical access is a problem.

> Driving FMC JTAG by MMC should be a default mounting option?

Since we don't have software written, and it's a specialized use case (no immediate demand to deploy Marble+CUTE-WR in the field), I vote to have the standard/initial build jumpered to give J14 access to FMC JTAG.

gkasprow, 2019-05-02:
> 
> 
>     The idea behind FMC JTAG was identical as we do in Sayma AMC.
> 
> OK. I haven't looked at Sayma design in months, I guess i should refresh myself.
> Can a remote in-the-field Sayma carrier, not attached to a bench-top JTAG pod, act as a JTAG bus master for the FMC JTAG chain?

In Sayma we do not have such a feature because there are other components in the ARTIQ environment that are updated via USB JTAG.

> There is a risk that FMC is not designed according to specification and the chain is not closed by default (TDO not shorted with TDI). In such a case the MMC should be able to overwrite the PRSNT [...]

not yet. One resistor and connection to the MMC does the job.

> 
>     Xilinx devkits have JTAG implemented in the same manner.
> 
> Can you be a bit more specific? Which devkit? Are you talking about the analog switch with an override feature? And is this the JTAG chain that is only accessible to a bench-top JTAG pod?

for example KC705. In our case, it is implemented in the same way.
Moreover, I would add 0R resistors that disable MMC JTAG.
MMC should have its individual SWD connector mounted by default. Using the same JTAG connector for both FPGA and MMC would be a real nightmare during development. I'm afraid that CPU debugger may have issues with talking with CPU with other JTAG devices in the chain. Another problem would be with SWD mode that won't work in such setup.
So I think dedicated SWD/JTAG connector is a must. Just copy working solution from Sayma and do not mount 0R resistors that connect it with rest of JTAG chain.

> MMC can use bit-bang JTAG, there is even Xilinx virtual cable protocol, but it makes sense when MMC has Ethernet access.
>
> The MMC is always assumed to have a data link to the outside world. In xTCA [...]

OK, this would work but I doubt if somebody finally implements it, having working self-programming solution :) Or are you talking about a link for diagnostics, sensor readout etc ?

> I don't see many use cases for MMC- FMC JTAG connectivity.
>
> OK. But I don't want to force people to add extra hardware in order to access the FMC JTAG chain. That hardware may be easy and convenient [...]

OK, It's better to add such support then.

> 
>     Driving FMC JTAG by MMC should be a default mounting option?
> 
> Since we don't have software written, and it's a specialized use case (no immediate demand to deploy Marble+CUTE-WR in the field), I vote to have the standard/initial build jumpered to give J14 access to FMC JTAG.

agree

michalgaska, 2020-09-15:
Can be closed

closed 2020-09-15 by michalgaska