BerkeleyLab/Marble-Mini #26: SW2 placement

ldoolitt, 2019-04-29:
SW2 looks ideally placed for AMC slot builds, but that's not very useful for NAD builds.
Please add a header in parallel, so a cable can be used to move this functionality to a chassis panel.  NAD builds would mark SW2 as DNP.

My ideal would be a 4x2 100-mil header with connections to two push-buttons (PB1 and SW2) and two LEDs (maybe with separate ballast resistors but otherwise parallel with LD11 and LD13?).  For height reasons this would have to be placed "to the right of" the FMC connector. Compatible switch/LED assemblies are cheap and widely available as part of ATX PC cases or case-substitutes.

gkasprow, 2019-05-02:
The idea behind SW2 was board reinitialization in the crate without a need for power cycling.
The use case for a stand-alone application would probably be different because power cycling is much easier than in MTCA.

ldoolitt, 2019-05-02:
Of course everything is software-defined, but my biggest interest in SW2 for a NAD is a "reset IP address to factory default" function.

gkasprow, 2019-05-02:
Good idea. We can also place tiny SMT switch available via panel hole with some needle.

ldoolitt, 2019-05-02:
> We can also place tiny SMT switch available via panel hole with some needle.

Fine with me, if you can make it work.  The panel side for a NAD is squeezed in by the Ethernet and SFP, and power entry, where there isn't a lot of room left.

gkasprow, 2019-05-02:
I use tiny switch that fits below Ethernet connector.
KSS241G LFS

michallgaska, 2019-05-09:
I added a tiny switch. 
@ldoolitt do you still need a header with connections to two LEDs?

ldoolitt, 2019-05-09:
> I added a tiny switch.

I see it, SW3.  It looks too close to the board edge.  Its drawing shows the button extending 0.8 mm from the pad; the layout in commit 7c5c3c37 only leaves 0.5 mm.

> do you still need a header with connections to two LEDs?

No, this is fine for now.

ldoolitt, 2019-05-10:
Re-opened as a reminder to shift the new SW3 (in parallel with SW2) left by at least 0.5 mm.

michallgaska, 2019-05-13:
Done

closed 2019-05-13 by michallgaska