BerkeleyLab/Marble-Mini #27: Review and organize DNP status for parts

ldoolitt, 2019-04-29:
We keep talking about parts being loaded for only one of the two styles of board assembly, AMC and NAD.  Since the first prototypes are headed for NAD-style testing at LBNL, AMC-only parts should be marked DNP (do not populate) for now.

We do need some organization and/or KiCad mechanism to clearly document what parts go with which build, such that a BOM and assembly file can be constructed cleanly for both types.  That step is _not_ a prerequisite for the first prototype (NAD) assembly.

gkasprow, 2019-05-02:
For the CERN project the Altium variants are not used for a number of reasons.
Each component has a field called "Assembled" and value "Yes" or "No". And this field is then generated with BOM, but NOT with p&p files!
This cames probably from an early version of Altium where assembly variants were not supported.
I remember the times where PCAD was used at CERN, then Altium purchased PCAD and discontinued it. All clients switched to Altium then. Maybe it is reminiscent of those times. 
Good assembly variant support like Altium has would be a great tool.
I tried to use Mentor Xpedition assembly variants but it was a real nightmare. The Altium approach where assembly variant information is preserved in a dedicated file is a good idea because it does not affect schematics nor PCB, only BOM and p&p files.

michallgaska, 2019-05-09:
I found a very nice tool to generate a BOM file. It supports PCB variants.
https://github.com/SchrodingersGat/KiBoM

ldoolitt, 2019-05-09:
> I found a very nice tool to generate a BOM file. It supports PCB variants. https://github.com/SchrodingersGat/KiBoM

Yes, that looks like it has the features we want.

Do you plan to set up for this (adding all the Config (fit_field) entries) before the prototype fab run?

michallgaska, 2019-05-10:
I made the first attempt to set up a variant for the standalone version using KiBoM. 
I have created _standalone_ variant of the board.
which clock is the default FPGA MGT clock? WR?

ldoolitt, 2019-05-28:
First prototype build will be standalone (NAD) style, using the WR clock.
Referencing commit d4c25c16, I think dnf attributes should be added to:
- [ ] S1
- [ ] SW2
- [ ] 16 capacitors routing MGT lanes to AMC fingers
- [ ] 4 capacitors routing MGT_CLK_1 to other than FPGA_REF_CLK0
- [ ] 2 capacitors routing MGT_CLK_0 to other than ??
I don't see a particularly useful source for MGT_CLK_0.  It might be nice to have an option for FMC2_CLK0_M2C, but that would conflict with the existing route of that clock to bank 34 MRCC pins.

That's all I see for now, but I'll keep looking.
@gkasprow  should also review this list.

Some of these, like the MGT lane routing, look already handled by the "config" attribute on the parts, which will drive KiBoM.  Other than clicking on the attribute editor for every part, do you know of a way to review all the "config" attribute values?

It would also be nice if there were documenation (in README.md?) for what the board's KiBoM options are, and how to generate the BoM file with that tool.  Maybe it's there, and I just don't see it.

ldoolitt, 2020-03-05:
The design as-built uses KiBoM as suggested, so I'll close this issue.

There is a minor residual problem that the build variant and resulting parts (or DNF) selection is not visible on schematics.  This is a generalization of #47, so that's where further discussion belongs.

closed 2020-03-05 by ldoolitt