BerkeleyLab/Marble-Mini #28: DDR3 schematic tweaks

ldoolitt, 2019-05-03:
A friend of mine looked through the schematic and gave me comments.
I hope this issue captures those ideas relating to the DDR3 schematic.

In general, deviations in this part of the schematic from Arty should be looked on with suspicion.
If we have a good reason to deviate, that should be noted and explained on the schematic.
I'm taking as reference Arty rev E.1, 2018-01-09
https://reference.digilentinc.com/_media/reference/programmable-logic/arty-a7/arty_a7_sch.pdf
7ab6ec142107efa10dd91654ad1f462544ba190af551292ac7d49c4649af9870  arty_a7_sch.pdf

Particulars:
- [ ] Change clock termination to 80 Ohms differential
- [ ] Change address termination resistors to 40 Ohms
(see https://www.xilinx.com/support/answers/42024.html, which talks about using 40 Ohms for the trace impedance, also)
- [ ] ODT pin termination should be 40 Ohms to V_tt
- [ ] Add more decoupling capacitors on +1V35, currently 8 where Arty has 19

gkasprow, 2019-05-03:
fully agree.
- [ ] there should be more capacitors on VTT. At least 5.
- [ ] 40Ohms are hard to get. use 39R9 instead.
- [ ] make sure that address lines impedance is 40Ohms.
- [ ] the clock signal is still referenced to VTT so I wouldn't use single 80R differential termination but 2x40R as it is now.

ldoolitt, 2019-05-03:
> 40Ohms are hard to get. use 39R9 instead.

OK, that's definitely close enough to 40. Arty uses 40R2, I rounded.

> the clock signal is still referenced to VTT so I wouldn't use single 80R differential termination

I see several clock signal termination topologies suggested by experts.  DIMMs use split termination resistors, center tap AC coupled to V_DD.  Weird. 
https://www.micron.com/-/media/client/global/documents/products/data%20sheet/modules/sodimm/jsf4c128x64hz.pdf
Of course, many experts also say to make sure to run signal integrity simulations as the final guide.

ldoolitt, 2019-05-03:
>>  40Ohms are hard to get. use 39R9 instead.

> OK, that's definitely close enough to 40. Arty uses 40R2, I rounded.

Searching Digi-Key for 1% 0402 resistors near 40 Ohms, the top 10 in quantity-available are all either 39R0, 39R2, or 40R2. I recommend putting 40R2 on our schematic.  There are just over 2 million Yageo RC0402FR-0740R2L on-the-shelf.

ldoolitt, 2019-05-09:
> [WP484 v1.0] uggests using 50R impedance and termination.

That document dates from September 2016.  AR# 42024 I gave above with the 40R suggestion is dated June 2014.  The Arty schematic that uses 40R2 is dated January 2018.

> we applied such DDR design in Kasli and it works perfectly.

At what speed?  The Arty also works perfectly.  A single resistor is easy to change, but a generic change from 50R to 40R means changing 25 such resistors, which would be tedious.

michallgaska, 2019-05-15:
I added more decoupling capacitors to the DDR3 power supply and VTT.
I changed the termination resistors to 40R2. 
I didn't change traces impedance.
According to the simulations, DDR3 will work.

[image]

closed 2019-05-15 by michallgaska