BerkeleyLab/Marble-Mini #31: SI issues

gkasprow, 2019-05-03:
the AMC connector signals have missing reference plane under some trace sectors
[image]

Moreover, the opposite AMC connectors contacts don't have any GND reference below.
The pad width is 0.47mm so to maintain 50R impedance, the reference should be placed 0.25..0.3mm below. Now there is no reference at all. Look how it is done on Sayma. On second layer there are cutouts in the plane. On L3 there is no copper
P2:
[image]
L3:
[image]

gkasprow, 2019-05-03:
The same applies to SFP pads. Look how it is done on Sayma
P2:
[image]

On L3 there cannot be any traces nor polygons below. To make sure I place there polygon cutout. 
[image]

michallgaska, 2019-05-06:
Fixed

closed 2019-05-06 by michallgaska