BerkeleyLab/Marble-Mini #32: Update addresses on I2C_MUX.sch

yetifrisstlama, 2019-05-09:
The embedded sketch with the I2C addresses seems out of date:

  * U39 is at 0x42
  * U35 is most likely at 0x50 (not really clear from the datasheet)

Also put a note that these are 8 bit I2C addresses (including the R/W bit = 0)

ldoolitt, 2019-05-10:
> U39 is at 0x42

Yes, that was fixed.

> U35 is most likely at 0x50 (not really clear from the datasheet)

I think it's really settable with its internal EEPROM, which is why the datasheet is non-specific.
I had some reason for listing 0x28, but maybe I got tripped up by by the I2C 7-bit address business.

> Also put a note that these are 8 bit I2C addresses (including the R/W bit = 0)

Good idea.

The source for this diagram is checked in to git as marble_i2c.eps, editable with xcircuit.
That file already has the U39 fix.

michallgaska, 2019-05-14:
Done

closed 2019-05-14 by michallgaska