BerkeleyLab/Marble-Mini #33: Test synthesis

ldoolitt, 2019-05-10:
Given the pin mapping file posted in #20 , take an exsiting useful design and synthesize for this board. Must happen before board goes out for fab, to make sure we haven't misunderstood some stupid Xilinx physical constraint.

ldoolitt, 2019-05-31:
Vamsi confirms RGMII and DDR3 are ok for synthesis.
FMC looks good, but not fully confirmed.  Still working on it.

ldoolitt, 2019-05-31:
@michalgaska please check DDR3 pin-compatibility with Xilinx MIG.

ldoolitt, 2020-03-05:
Ethernet, MMC, I2C, and Pmod connectivity all work as expected on hardware. FMC connectivity passes tests using the OHWR [FMC Carrier Tester board](https://ohwr.org/project/fmc-conn-tester/wikis/home).  I think the only thing left to do before closing this bug is DDR3 checkout / confirmation. @michalgaska, any progress on Xilinx MIG?

michalgaska, 2020-03-06:
Unfortunately, I don't have any progress with MIG.

ldoolitt, 2020-09-12:
Closing, since the first-prototype run (commit a1ed1ff17) has proven itself with usable with designs from:

* [Bedrock](https://github.com/BerkeleyLab/Bedrock) (no DDR3 support)
* [LiteDRAM](https://github.com/enjoy-digital/litedram) (using all 1024 GiB DDR3)
* conventional Xilinx uBlaze/MIG (but only with 512 GiB DDR3 visible)

closed 2020-09-12 by ldoolitt