BerkeleyLab/Marble-Mini #34: Include revision date and git commit ID on board

ldoolitt, 2019-05-10:
Place-holder issue to remind @michalgaska to put board design revision info (including git commit id) on something (silk, copper) physically visible on the hardware.  Necessarily done at the last minute, after all other changes.

Ideally this step will be automated, but that's definitely not essential for the first prototype.

csp, 2019-05-30:
Can we include a version number corresponding to a tag? The first revision should have v1.0 printed on the board, this makes it easy to identify them when we go into production of the next versions and end up with boards of different production runs

michalgaska, 2019-11-13:
I think it is done.

jersey99, 2020-02-04:
Closing this.

closed 2020-02-04 by jersey99