BerkeleyLab/Marble-Mini #37: Symbol library management

yetifrisstlama, 2019-05-29:
Larry found some issues related to ERC in some of the schematic symbols. However as long as we load the symbols from the `-cache.lib` file, we cannot edit them.

I suggest to rename `AMC_FMC_Carrier-PcbDoc-cache.lib` to `AMC_FMC_Carrier-PcbDoc.lib` and include it in Kicad as a project specific symbol library.

I found [this guide](https://hackaday.com/2017/05/18/kicad-best-practises-library-management/) explaining the details.

Ideally we should not have to depend on `-cache.lib` or `-rescue.lib`

ldoolitt, 2019-06-13:
This specific problem is the tip of a larger iceberg.  We really need the library entries for the specific parts this design uses, in order to maintain or upgrade this design going forward.  This includes getting access to the 3D model part of these items, so as to be able to produce a 3D rendering of the assembled board as shown on this project's README.  This also practically prevents us from pushing changes to github, since every time we load the design, Kicad redirects library links to the local cache.

As I understand it, most or all of these parts come from a CERN library, and there are negotiations taking place for how and when to copy. license, and publish this subset.

This bug sounds trivial, and definitely shouldn't slow down our production of prototypes, but is a critical bug moving forward!  Until it is solved, the OHWR nature of this design is hollow.

ldoolitt, 2020-02-02:
Some WIP on this issue is now in the local-lib branch.
Tomasz is also looking into it.

ldoolitt, 2020-05-05:
More progress in the fab2 branch, based on great work by @yetifrisstlama .
Specifically as of commit 6cafe828 the ERC fixes are in the self-contained libraries.
3D models are **not** currently available.
I'd like to verify that features can be added in this mode before closing this issue.

ldoolitt, 2020-05-27:
Still missing the 3D models, but the fab2 branch feels maintainable, with coordinated commits from three people.  Closing this issue.

closed 2020-05-27 by ldoolitt