BerkeleyLab/Marble-Mini #46: Required hardware patches to the first version of Marble

michalgaska, 2019-11-13:
- USB FTDI - reset pin connect to VUSB
- remove R225
- remove F1 and F2
- rotate LEDs
- FPGA JTAG connector pin connect to P2V5
- wrong transistors, BSS139 -> BSS138
- change S25FL128S to S25FL128SAGMFIR01

yetifrisstlama, 2020-05-02:
> USB FTDI - reset pin connect to VUSB

It is connected to VUSB right now, where should it connect to instead?

ldoolitt, 2020-05-02:
Michael -

On Fri, May 01, 2020 at 06:16:26PM -0700, M. Betz wrote:
> > USB FTDI - reset pin connect to VUSB
> It is connected to VUSB right now, where should it connect to instead?

At least some of thse bugs may have been fixed in

commit ea1216ff2fbc74d5afe56203d6e5192d8b450fb0
Author: michalgaska <michalgaska2@gmail.com>
Date:   Wed Nov 13 08:16:37 2019 +0100

    applied all required amendments to the project

  - Larry

yetifrisstlama, 2020-05-02:
> remove F1 and F2
These are still `FUS0603_BOURNS-SF-0603F050-2`. Shall they be replaced by 10 Ohm resistors? Which footprint?

> rotate LEDs
Not sure how to confirm that one, maybe Michal can comment if he fixed it.

I could confirm all others have been fixed.

ldoolitt, 2020-05-30:
We've also learned that the PM I2C bus (attached to the microcontroller) is missing pull-ups.
Fixed in the design in commit 078b7900.
Patching hardware is possible with clip-leads using Pmod for the +3V3, and tying in to SCL and SDA using J9.

ldoolitt, 2020-06-24:
Also see issues #59 (Fan connectors not functioning) and #60 (Annoying MMC reset behavior).

closed 2020-09-15 by michalgaska