BerkeleyLab/Marble-Mini #47: Unclear which capacitors are populated on MGT_CLK_*_P/N lines

jersey99, 2020-02-28:
The schematic doesn't provide any information to the user about which of the capacitors on the lines are populated.

I can confirm that C296 and C297 are placed, because I can drive the transceiver with 125MHz clock on FPGA_REF_CLK1_P/N

Sidenote: It is slightly confusing that FPGA_REF_CLK1 goes to MGT_REF_0 .. Not a big deal I suppose.

ldoolitt, 2020-03-06:
With enough clicking on the live schematic (not a PDF rendering!) you can find out how components are tagged for build options, and figure this out based on the [KiBoM](https://github.com/SchrodingersGat/KiBoM) documentation.  I agree, this is too indirect to be considered "clear".  Maybe we shouldn't try to fix this ourselves, without help or at list advice from the KiBoM community?

ldoolitt, 2020-05-27:
Fixed as of commit 078fe5ad.  Thanks @yetifrisstlama!

closed 2020-05-27 by ldoolitt