BerkeleyLab/Marble-Mini #48: Updates targeting next fab run

ldoolitt, 2020-03-06:
If any of these are complex or controversial and need discussion, maybe start a separate issue and remove them from this checklist.  This list may not be complete, because some features of the board related to the MMC have not yet been tested.  I claim that ignorance shouldn't hold back making the changes we know about.

Clear errata hand-fixed on prototypes:
* [ ] USB FTDI - reset pin connect to VUSB
* [ ] mark R225 as DNF
* [ ] replace F1 and F2 with 10 Ohm resistors
* [ ] rotate LEDs
* [ ] FPGA JTAG connector pin connect to P2V5
* [ ] wrong transistors, BSS139 -> BSS138
* [ ] change S25FL128S to S25FL128SAGMFIR01

Improvements:
* [ ] add direct-attach LED or two to FPGA, maybe one on interrupt line to MMC?
* [ ] add on-board SiLabs oscillator as option for MGT_CLK_0
* [ ] add 330 Ohm series resistor to all four FMC GA pins
* [ ] add physical power switch (includes barrel and PoE but not AMC?)
* [ ] add power meter (another IN219) on 12V bulk input
* [ ] move parts between sides to reduce number of reels used in loading (moving 13 components could save 13 reels: C2 C42 C374 D1 D2 Q2 Q19 R58 R66 R103 R104 R108 R160)
* [ ] U4 (88E1512 Ethernet PHY): VDDO_SEL (10) should be tied low for VDDO = 2.5V
* [ ] move U29 (I2C thermometer) to other end of board, near VCXOs
* [ ] remove stray via at (272.35, 172.53)
* [ ] more (labeled) test points, especially power supplies

Improvements that don't actually change copper:
* [ ] NAD builds can mark S1 as DNF
* [ ] AMC builds should mark Pmod J12, J13, J16 as DNF
* [ ] mark SFP TX_DIS pulldown resistors as 0 Ohm DNF: R84 R105 R185 R187
* [ ] update branding/naming on silkscreen
* [ ] more labels in general, e.g., DONE by associated LED
* [ ] much bigger silkscreen labels for user-facing parts: J9, J10, J11, J12, J13, J14, J15, J16, S1, SW1, SW2, SW3, M1, M2
* [ ] schematic: clarify/unify net names +3.3VP (power), 3V3PM (AMC), +3V3MP (many)
* [ ] increase voltage rating of capacitors attached to +12V bus to 25V (some are 16V now)

yetifrisstlama, 2020-05-01:
More wishlist items collected from `marble1/todo`

- [ ]  add rubber feet (with designated location) for desk use
- [ ]  add mounting holes (standoffs)? (overlap with rubber feet)
- [ ]  use 10 Ohm instead of F1 and F2
- [ ]  reduce BOM line count in general, push-buttons in particular
- [ ]  if keeping J14 "Cortex Debug Connector" make a big pin-1 marking can we use a keyed/shrouded option for J14? Amphenol 20021221-00010C4LF
- [ ]  improve testability in general
- [ ]  improve documentation of build options

ldoolitt, 2020-05-07:
As of fab2 branch commit 9ce94581, I have moved seven parts (C374, D1, D2, R58, R66, R103, R160) between sides, to cut back the number of reels needed in assembly.

One group of tasks, that should at least be studied together, is to rework the +12V bus:

* add physical power switch (includes barrel and PoE but not AMC?)
* add power meter (another IN219) on 12V bulk input
* move parts between sides, the perfect-diode circuit from barrel J1: Q1 Q2 R104 R107 R108

After that, the only remaining parts that I want to move are C2 C42 Q19.

ldoolitt, 2020-06-01:
Regarding the @yetifrisstlama comments above: I just ticked off two.

* Reduce BOM count, I shaved it from 128 to 117.  I'm sure it can go lower, but only with a lot more work.
* improve documentation of build options, see the newly visible config attributes and associated documentation on FPGA_MGT.sch.

That leaves Improve testability in general, which I left open. The new 11 test points on power supplies should help some.  If anyone has specific ideas, please comment here or in a new issue.

ldoolitt, 2020-06-02:
Thanks for everyone's persistence and patience!  Closing.

closed 2020-06-02 by ldoolitt