BerkeleyLab/Marble-Mini #54: mini j-link segger connector

jersey99, 2020-05-19:
It's hard to know which orientation to plug in the segger j-link mini onto the Marble-mini. It would be good if some hints are added, possible options:
1. Add a mating connecter with the segger j-link notch
2. Add a note on the silkscreen. I understand that J16 mark on the silkscreen gives orientation, but a little more indication on the silk screen would help double guessing and having to look at the docs every once in a while.

@ldoolitt @yetifrisstlama this would be a "good to have" before the next spin.

yetifrisstlama, 2020-05-19:
Assuming you mean J14 (MMC JTAG)

This doc about [cortex debug connectors](http://infocenter.arm.com/help/topic/com.arm.doc.faqs/attached/13634/cortex_debug_connectors.pdf) suggests it should be a [`FTSH-110`](http://suddendocs.samtec.com/catalog_english/ftsh_mt.pdf) connector.

Looks like we can just install the shrouded version, if we can come up with the right part number.
FTSH - 110 - ???

Here's some options on digikey which might fit too:
https://www.digikey.com/short/z4fr7m

jersey99, 2020-05-19:
Sorry, yes. To confirm J16 (MMC JTAG).

They all look good to me. :-|

ldoolitt, 2020-05-19:
In the comment added to Issue #48, this shows up as

* if keeping J14 "Cortex Debug Connector" make a big pin-1 marking can we use a keyed/shrouded option for J14? Amphenol 20021221-00010C4LF

This in turn was pasted from the to-do file in the marble1 repo.  So yeah, it's a known issue.
Changing assignment to Michal, since he agreed in a phone call to at least look into making this change.

yetifrisstlama, 2020-05-20:
Note that `20021121-00010C4LF` is not shrouded

I'd recommend: `Samtec FTSH-105-01-L-DV-K-TR`

ldoolitt, 2020-05-20:
> Note that 20021121-00010C4LF is not shrouded

Looks shrouded to me.
At [Digi-Key](https://www.digikey.com/product-detail/en/amphenol-icc-fci/20021221-00010C4LF/609-3700-1-ND/2209154).
This is the part inserted by Michal, which now shows up in fab2 branch.
Both the library and the instance need the SMD attribute set to Yes.

yetifrisstlama, 2020-05-20:
For the `20021121-00010C4LF`, how do we know the notch in the shroud fits the connector? It's hard to tell from the drawing if there is a notch at all.

gkasprow, 2020-05-20:
In such cases I download 3D models and check with FreeCad, Alibre or similar tool if they fit.

yetifrisstlama, 2020-05-20:
Excellent tip! 

Looks like there's no notch on `20021221-00010C4LF`

[image]

ldoolitt, 2020-05-21:
Right.  Shrouded but no notch.  The connector shipped with the Segger demands a notch.
The silkscreen for the footprint should also show the notch, so its chance of getting assembled right is more than 0.5.

The FTSH-105-01-L-DV-K-TR you mentioned earlier looks a little funny, and is three times the price, but will probably work.

yetifrisstlama, 2020-05-21:
The `FTSH-105-01-L-DV-K-TR ` is the debug header recommended by ARM. I guess that's also why it's :moneybag::moneybag::moneybag:

ldoolitt, 2020-05-22:
FTSH-105-01-L-DV-K-TR placed as of commit f78131c.

closed 2020-05-22 by ldoolitt