BerkeleyLab/Marble-Mini #56: Keep some I2C busses running with only +3V3P

ldoolitt, 2020-05-23:
Even when the "main" power is shut off with EN_PSU_CH or OVER_TEMP, the microcontroller should be able to run basic services like sense temperatures and read FMC ID EEPROM.

As of now (commit d5d2528) this is partially implemented, thanks to @michalgaska and particularly his addition of U46 TCA9517.

Reading FMC ID EEPROM requires the bus mux U5 and the FMC 3V3VAUX to be powered by +3V3P, which is done.  Still to-do

* [ ] Move pull-ups R45 R46 R47 R48 to +3V3P

Keeping the I2C_APP bus running, so the microcontroller can see the input voltage (presumably its current is close to zero), requires

* [ ] Change U34 U39 (PCA9555) supply to +3V3P

Want to check other things, like

* [ ] Should I2C_FPGA_SW_RESET and EXP_INT be isolated from the FPGA like SCL and SDA?

gkasprow, 2020-05-23:
One can use such a circuit to isolate the I2C busses
[image]

ldoolitt, 2020-05-23:
@gkasprow, can you elaborate on how to use that circuit?  I can try to reverse engineer it: It looks like EEM0_I2C is the always-on side, and the I2C0_{SDA,SCL} side can be left powered off if P12V0A is "low".  Turning this gate on requires 12V0A that is between 2 and 20 V higher than P3V3. With some work, we could probably create such a control signal for our board.

gkasprow, 2020-05-23:
These are two independent circuits. The right side is always on, left is connected when the main power is available. 12V makes sure that both FETs are open both for L and H. Even with the left side shorted to GND, the right side still works, because diodes in FETs are not conduction. Of course, the right side needs pullups to P3V3_MP

ldoolitt, 2020-05-25:
@gkasrpow, I must be missing something about your explanation.  If the left side is shorted to ground, I see protection diodes in each N-FET that will pull the right side down.  If the left side "just" has protection diodes to ground, maybe that's OK, the right side can get pulled up to around 1.4V.

Is this cheaper/faster/smaller than the TCA9517? That part has an under-voltage lock-out, plus an enable input (not yet used, but could be plausibly be hooked to a P_GOOD signal from the "main" 3.3V supply) .

gkasprow, 2020-05-25:
Yes, this TCA is a better option. I used transistors because I didn't want to add another item to the BOM. The transistor option isolates the only one side and was used to disconnect part of the I2C rail supplied from 3V3. If the left side is shorted (by non-powered 3V3 which is derived from non-powered 12V) nothing happens because substrate diodes are not conducting and FETs are in the off state. When 12V is present, 3V3 rail is present as well.

ldoolitt, 2020-06-01:
I'm marking this done as of commit 8f30f0a6, where a second TCA9517 has been added to isolate I2C_FPGA_SW_RST and EXP_INT from an un-powered FPGA. If anyone reviewing the schematics discovers something we missed, please re-open this issue. Of course it will remain to be verified in hardware.

closed 2020-06-01 by ldoolitt