BerkeleyLab/Marble-Mini #57: Missing I2C pull-up resistors on I2C PM bus

smpaiagua, 2020-05-27:
There are pull-up resistors on the IPMB and FPGA I2C busses but not on the PM bus, making it inaccessible from the microcontroller. This issue was encountered when checking out I2C connectivity in one of the original Marble Mini boards.

A temporary workaround is to use the J9 connector to clip on the necessary pull-up resistors.

ldoolitt, 2020-05-27:
Fixed for the next rev in commit 078b7900.  Let's leave the issue open until at least one of the existing prototypes is patched and confirmed working.

ldoolitt, 2020-05-28:
PM I2C bus is demonstrated to work when patched with pull-ups, so the fix is hardware-proven.
I'm closing the issue, even though we have five un-patched prototypes out there.

closed 2020-05-28 by ldoolitt