BerkeleyLab/Marble-Mini #59: Fan connectors not functioning

ldoolitt, 2020-06-15:
Fan connectors M1 and M2 don't spin fans. I don't claim to fully understand the situation.
I'm pretty sure pins 1 and 3 are swapped, but in my limited testing with that swap un-done, the fan still didn't spin.
It looks like U27 MAX6639 comes up by default at 50% duty factor.  In order to monitor and adjust its behavior on the prototype boards, you need to work around issue #57 (not so hard with clip leads and thru-hole resistors).  We have proof-of-concept MMC code that can manipulate this chip over I2C at  https://gitlab.lbl.gov/spaiagua/marble_mmc.

gkasprow, 2020-06-15:
you can have a look at Booster design where this chip works.
https://github.com/sinara-hw/Booster/releases
Main Board design, page 13

ldoolitt, 2020-06-18:
I have a report from @michallgaska  indicating that pins 1 and 3 are indeed swapped in the current (commit bbb0a2e2) design.  Un-flipping those pins makes a fan work.

ldoolitt, 2020-06-24:
Fixed in the fab2 branch in commit 0453e1eb.
Existing prototypes could be reworked by rotating the fan headers 180 degrees.

closed 2020-06-24 by ldoolitt