BerkeleyLab/Marble-Mini #61: FPGA over Network Update

gkasprow, 2020-07-23:
We are [working on this featur](https://github.com/elhep/AMC-FPGA-Network-Update/issues/1)e for our FPGA boards.
Would be nice to join forces since the idea is nearly identical

ldoolitt, 2020-07-23:
That URL 404's for me.  Typo?  Permissions?

marmeladapk, 2020-07-23:
Permissions. @gkasprow, the repository is set as private. You have to grant access.

gkasprow, 2020-07-23:
@kaolpr can you make it more open, please ?

kaolpr, 2020-07-23:
Done. Forgot to change after setting things up.

kaolpr, 2020-07-23:
Do you have any work already done in this subject?

ldoolitt, 2020-07-23:
Yes, I posted to the AMC-FPGA-Network-Update issue.

ldoolitt, 2020-09-11:
I see no reason to leave this issue open, since we have established contact and are open to sharing.
My guess is that AMC-FPGA-Network-Update is headed for a heavier-weight paradigm than we want to include in baseline board-support for Marble-Mini.  Of course the hardware is agnostic.

closed 2020-09-11 by ldoolitt